bigtreetech/BTT-SD-TF-Cloud-V1.0 #1: password length limited?

brutzler, 2020-06-01:
Does anybody know, if the password length is limited?
My SD-Cloud gets a 
```
Delay...
Reading failed!
Fatal exeption 28(LoadProhibitedCause):
```
when I use my "normal" password with 63 characters.
Changing the password of my Home-WiFi to a shorter one (10 characters), I get a normal WiFi connection.
Maybe a too small Array-decleration?
But without Source code I can only guess

oldman4U, 2020-06-08:
Hi brutzler.

May i ask you a question and it is not about the length of your password.

Using the Cloud device, can you:
1. Upload gcode files directly from Cura
2. Can you also upload firmware.bin files and update the firmware of the board of TFT  after that

Thank you und einen schönen Tag

Evg33, 2020-06-08:
https://github.com/ardyesp/ESPWebDAV

brutzler, 2020-06-11:
> 
> 
> Another sources:
> https://github.com/ardyesp/ESPWebDAV
> https://cloud.mail.ru/public/3zqi/3ecDR82hg

What sources are this?
Is this the original source from bigtreetech? Or did bigtreetech only copy some stuff from the internet. I do not want to change the firmware on the device without any "way back"

brutzler, 2020-06-20:
Looks like the second link [https://cloud.mail.ru/public/3zqi/3ecDR82hg] is the software bigtreetech is using for their modules.
And there I can see, that the array-length of the password is defined with "only" 50 characters.
What I expected. I will modify this when I have more time and post the results.

Finally to say: The help of bigtreetech for this issue was very lousy. In my opinion they output more and more hardware stuff and forget that support is part of business.

oldman4U, 2020-06-22:
Hi brutzler.

This is Open Source software and BTT is making the hardware. The success of the community depends on the users which are trying helping other users by answering their questions (for example my questions above which you ignored so far;-)).

Please do not forget to close this ticket in case you do not need it anymore

Danke

brutzler, 2020-06-22:
Hi oldman4U,
imho its a little bit different. BTT takes free sources to make profit. Without the software from the community the module will never work.
But to be honest. Its the same as with SKR1.3 and marlin. 

To your questions. I did not ignore them. I could not really answer them as long as the module was not working.
1. I do not use Cura. But it is possible to upload gcode directly from S3D. Why not from different slicers. The module is working as network drive in my W10-computer. I do this since months with a Toshiba flash air. And this module is working the same way.
2. Upload bin files. This is possible as well. And why not flashing the firmware of the TFT.  The module is working in the SD Slot as a normal SD-card.
The only difference to a normal SD is that you do not have to put it into the cardreader of the computer to copy files on it. This is done over Wifi.

Hope i could answer your questions sufficient.

oldman4U, 2020-06-23:
Hi brutzler.

Thank you for answering my questions, i really appreciate. Based on your answers i believe it is the right product for me, the price is more than fair and it should work with my other hardware. So i will buy it.

Thats the beauty of a free world, we can have different opinions and hopefully respect the other. This free world also allows to choose, so i would be interested to know from you, why you have chosen BTT over all other solutions in the market? Would be great if you could let me know.

Thank you again

brutzler, 2020-06-24:
Hi oldman4U,

Why BTT?
I already have a SKRv1.3 in my printer. Working great with my TFT3.5 (BTT too). 
A SKR V1.4 is waiting for use.
And a friend told me he found a cheap alterative to the Toshiba flash air. And for that price, I bought.

But to say: BTT is not my first choice, if you need support. They are friendly, I got always an answer (sometimes days delayed), but its a first level support. Sometimes I think, they do not really understand the problem, or I am to stupid to explain it right.

What other solutions for a WiFi-SD-Card do you know?

oldman4U, 2020-06-24:
Hi brutzler.

So it seems that you had some BTT hardware already and you bought even more. To me, this sounds like you like their stuff, even it does not answer my question why you finally bought it and not a black box solution from another vendor with better support.

Duet WIFI for example. I am sure their solution works better out of the box and they are not squeezing out the open source community to become rich. So you get everything you ask for and you will not have to complain about several things.

brutzler, 2020-06-26:
Duet Wifi is a complete board. I already have a running board (SKR V1.3) working fine. No need for change the whole board.
And you have to pay a lot of more money for it. But this is normal: Better support causes higher price. (If the support of Duet is better, I do not know.)
I was looking for an alternate module to my Toshiba FlashAir. Therefore I bought it. And because I am always interested to play with new hardware :-)
And finaly I got it working so far. 
Hope you have the expected success with your SD-Cloud!

oldman4U, 2020-06-26:
Maybe your next board is a Duet and you let us know how it works;-) I also like it to tinker and learn, so I bought the SD-Cloud and hope it works properly.

Please do not forget to close this ticket, in case you think you do not need it anymore.

Thank you and happy printing

knive87, 2020-06-28:
Hi friends! Please help me. BTT TF Cloud does not work. Bought on aliexpress BTT SD Cloud. I made all the settings as on video, but ESP does not see the card, or the problem is in the firmware. All connections from ESP to the card rang, everything is OK. But in the terminal it just writes AI Technologies and Ready. Manipulations with the card (replacement with another, formatting in FAT16, FAT32 do not help). Maybe someone came across? You can of course try to re-flash ESP yourself, but for now everything is under warranty.

oldman4U, 2020-06-28:
@brutzler 

Could you please close this ticket. ( and maybe help knive87);-)

@knive87 Please open your own ticket instead of using this one which covers another question and which is already answered and should be closed therefore 

Thank you

closed 2020-06-28 by brutzler

knive87, 2020-06-28:
Ok. I do

> 
> 
>Closed  #1 .
 
 
С уважением,
Антон Зайкин
knive87@mail.ru