bigtreetech/BTT-SD-TF-Cloud-V1.0 #2: Not Connect without usb

giorgio64, 2020-06-11:
connect onli if powered from usb cable, but if inserte on reader, not work

oldman4U, 2020-06-22:
Hi giorgio64.

Does the reader work in case you use a standard SD/TFT card?

giorgio64, 2020-06-22:
Hi, don't work. I try all possible but don't work... I can connect, i can send files, but the reader don't read inside the sd

oldman4U, 2020-06-23:
Hi.

Sorry, but my question was, if the card reader works properly in case you put a standard SD card into it, not the Cloud.

giorgio64, 2020-06-23:
Yes my reader with standard sd card work fine

oldman4U, 2020-06-23:
OK.

Which TFT or Mainboard are you using with the device and do you have another which you could use for a test?

oldman4U, 2020-06-25:
Hi.

Any news regarding the reported issue?

curso007, 2020-07-10:
Dear,

I can confirm that without being connected via USB, it does not connect to the wifi!
When I connect to a power bank or my PC via US, I can transfer files but what is the point the aim was to do the same but when inserted in the SD slot of the printer!
For information I even flashed the card with the latest firmware available here, from June 30th I think with the same result!
It is really disappointing. Especially that it seems that the average speed can reach 250KB/s, that was my minimum hope.
I hope that this a software issue that be fixed ASAP.

Br,
Cataldo

aksaek, 2020-07-12:
i have the exact same problem. Bought the WIFI-Module with the SKR Mini E3 V2.0 for my Ender, which worked fine after installing. Today I set up the module, had access from the pc, but in the sd-card slot its just randomly blinking

geraldjust, 2020-07-16:
the problem it seems it might not have enough current on most printers 3.3v line to power the esp8266. so when giving it more juice from a external 5v supply to the internal 3.3v regulator then it can do it.

Evg33, 2020-07-21:
Not working with BTT SKR 1.4Turbo :(

NAPCAL, 2020-08-28:
Just found this out on my BTT Pro V1.1 using Marlin 2.0.6.
Configuration_adv.h 
When selecting #define SDCARD_CONNECTION ONBOARD (Configuration_adv.h), the TF cloud device will not start the Blue led blinking (when trying to connect to WIFI) and as a result, can not connect to the device via computer for file transfer.

By changing this selection to #define SDCARD_CONNECTION LCD (Configuration_adv.h) will allow the TF cloud to connect for usage.

nigelpayres, 2020-09-12:
I have the same problem with the SDCard Cloud. When it's plugged into my PC via USB it connects to my WiFi without problem. If I plug it into the (working) SD card slot on my TFT35 V3.0 it does not. There is no WiFi connection at all, although the files on the card are visible on the TFT itself

nigelpayres, 2020-09-12:
> i have the exact same problem. Bought the WIFI-Module with the SKR Mini E3 V2.0 for my Ender, which worked fine after installing. Today I set up the module, had access from the pc, but in the sd-card slot its just randomly blinking.
> 
> Edit: if connected via USB, I can ping the module, via SD-Slot I can't

Doesn't the documentation say that it doesn't work on the V2.0 TFT as it uses SDIO rather than SPI?

3d-designs, 2020-09-14:
> Dear,
> 
> I can confirm that without being connected via USB, it does not connect to the wifi!
> When I connect to a power bank or my PC via USB, I can transfer files but what is the point the aim was to do the same but when inserted in the SD slot of the printer!
> For information I even flashed the card with the latest firmware available here, from June 30th I think with the same result!
> It is really disappointing. Especially that it seems that the average speed can reach 250KB/s, that was my minimum hope.
> I hope that this a software issue that can be fixed ASAP.
> 
> Actually the card is working for now only as an SD adapter 8(.
> 
> Br,
> Cataldo

May I ask how you managed to flash the firmware? I assumed it's just a case of uploading to the root of the inserted card, but that doesn't seem to do anything here.

oldman4U, 2020-09-14:
Please check this. 

https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/issues/6#issuecomment-653908175

3d-designs, 2020-09-14:
> Please check this.
> 
> [#6 (comment)](https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/issues/6#issuecomment-653908175)

Thank you. I am looking at it now.

3d-designs, 2020-09-25:
Did anyone manage to resolve this problem?

oldman4U, 2020-09-26:
Do you have the same problem?

oldman4U, 2020-09-26:
My new Read Me is online. So I hope this issue will come up less often. But there seems to be a problem with some precompiled firmware. Do you know how to fix it, radek?

3d-designs, 2020-09-26:
> Do you have the same problem?

Yes, I can connect to the WiFi absolutely fine when it's connected via USB to my PC. When it's plugged into the TFT35 there is no WiFi. The SD works (that is, the SD card contents are visible on the TFT) but WiFi doesn't start.

oldman4U, 2020-09-27:
And you have a red and green led at the button lit up and is the blue Led flashing every 20-30sec?

No whitespaces in the SSID name?

3d-designs, 2020-09-27:
> And you have a red and green led at the button lit up and is the blue Led flashing every 20-30sec?
> 
> No whitespaces in the SSID name?

It's not an SSID issue as the WiFi connects fine when powered by the USB. When it's connected to the TFT the LEDs seem to be working, but the device won't connect to the WiFi, even though the TFT is only 1m away from the USB socket on my PC, so it's not a range problem either. 

If I move it back to the PC, it works normally.

oldman4U, 2020-09-27:
I asked something different.

3d-designs, 2020-09-27:
> I asked something different.

I don't understand what you mean.

oldman4U, 2020-09-27:
And you have a red and green led at the button lit up and is the blue Led flashing every 20-30sec?

3d-designs, 2020-09-28:
> And you have a red and green led at the button lit up and is the blue Led flashing every 20-30sec?

No, what I see is

All LEDs flash very briefly on power on
Red and Green LEDs stay on
After about 10s the blue one flashes furiously for about 5-6 secs but does not come back on again after this

Ah... althought very oddly indeed, it is now actually connecting, despite my not having changed anything. I had tried it countless times before and now it seems to have worked. That's worrying.

3d-designs, 2020-09-28:
> And you have a red and green led at the button lit up and is the blue Led flashing every 20-30sec?

Ah, I think I now have more data:

If I connect the WiFi to the TFT when it's connected to the SKR, it doesn't work and behaves as I describe. This includes flashing the blue LED every 20 secs or so. If I reset everything, the WiFi starts to work as it seems to connect _before_ the TFT connects to the printer. If I remove and replug it while connected, it doesn't work.

oldman4U, 2020-09-28:
Moving target. 

Mine has red and green on as soon as there is power. Blue flashes also followed by approximately 25sec nothing, then the blue flashes 41 times and I can connect after that. Once I have the time I will make a ticket and describe all those things.

Let me know how it works for you.

oldman4U, 2020-09-28:
If I connect the WiFi to the TFT when it's connected to the SKR, it doesn't work and behaves as I describe.
You mean you start everything and then you put the WIFI into the TFT. I saw this also, but finally this is not a real life case - at least for me. Or do you mean something else?

Also. when i remove the SD card with the SETUP.INI and start the printer, the WIFI module shows the blue LED flashing every 20sec or so. This seems to be a timeout it tries to connect to the WIFI in case no connection could be established. I believe this even more, because when I have this behaviour and I insert the SD card, the next cycle, the module connects to the WIFI.

Another thing: SETUP.INI has to be plain text. Using a RTF textile does not work.

oldman4U, 2020-09-28:
If I connect the WiFi to the TFT when it's connected to the SKR, it doesn't work and behaves as I describe.
You mean you start everything and then you put the WIFI into the TFT. I saw this also, but finally this is not a real life case - at least for me. Or do you mean something else?

CORRECTION: Inserting the WIFI module once the printer is already powered up - once the WIFI module reaches a timeout interval, the blue LED of the WIFI module flashes 41 times and connects to the WIFI.

In any case, once connection has been established, the every 20sec flashing does not happen anymore and the blue LED only flashes during read/write operation.

3d-designs, 2020-09-28:
> If I connect the WiFi to the TFT when it's connected to the SKR, it doesn't work and behaves as I describe.
> You mean you start everything and then you put the WIFI into the TFT. I saw this also, but [...]

I mean that it seems as if: If I connect the WiFi when the TFT has a connection to the printer, the WiFi doesn't want to connect to the WAN. If I connect the WiFi when the TFT isn't connected (on power on, or a reset) then it will connect. I have left the device connected for quite some time in the first scenario and it doesn't seem to connect however long I seem to leave it.

oldman4U, 2020-09-28:
I used 2 different mainboard (SKR v1.4 and SKR Pro v1.1) and 2 different display (TFT35 E3 v3 and TFT28) and they all work, does not matter how I connect them.

The TFT has to be connected to the mainboard during startup, everything else makes no sense.

So you have the mainboard connected to the printer, but the WIFI does work, what does the blue LED at this time?

3d-designs, 2020-09-28:
> I used 2 different mainboard (SKR v1.4 and SKR Pro v1.1) and 2 different display (TFT35 E3 v3 and TFT28) and they all work, does not matter how I connect them.
>
> The TFT has to be connected to the [...]

Sorry, I was ambiguous. When I say not connected, I didn't mean physically, but logically. When the printer starts, it take a little while until the TFT detects and connects to it. In this state, the WIFi will connect. If the connection is already there, then the WiFi will not work. If I unplug and re-plug the WiFi it will not connect. If I then reset everything (press the reset button) it will connect. Similarly, from a cold start it seems to connect OK. If it loses connection then the only way to resurrect it is to reset everything.

oldman4U, 2020-09-28:
So you mean that everything works fine when starting the printer, but once you lose connection, because you accessed the SD card from the TFT, you can not access the SD card anymore but you have to restart the printer. Right?

Or do you mean something else?;-)

3d-designs, 2020-09-28:
> So you mean that everything works fine when starting the printer, but once you lose connection, because you accessed the SD card from the TFT, you can not access the SD card anymore but you have to restart the printer. Right?
> 
> Or do you mean something else?;-)

Yes, I suppose that's what it boils down to. Originally I couldn't seem to access at all when it was connected to the printer rather than the USB. However, I discovered this morning that it does seem to work immediately after a reset for at least a while. If it then becomes inaccessible (which seems to happen frequently) then it's not possible to do anything other than reset the whole printer. Simply removing the WiFi and re-inserting it (essentially resetting it and not the printer/TFT) doesn't work. It seems to try to connect, but is unable.

oldman4U, 2020-09-28:
I agree and finally it is a shame how this "works".

Baltojikale, 2020-09-29:
I have BTT SD Cloud v1.0. Same problems here. with usb power everything works as it should but on tft35 e3 v3 no wifi connection - acts as micro sd card reader. Been checking this post and manual of device. Seems that timeout vs resenting of tft could be related to spi and sdio connections... Manual of sd cloud states that it only works on spi mode not sdio . Could this be that TFT by default initially while not connected to mainboard has open spi and after connection spi is busy thus sd cloud cannot connect??? Thus could it be that tft fimaware need to be updated ??? ill be checking that...

oldman4U, 2020-09-29:
Connect it to the TFT, start the printer and check what the blue LED does
the first two minutes.

Let me know

oldman4U, 2020-09-29:
Please update the firmware and let me know.

oldman4U, 2020-09-30:
@Baltojikale 

This is related to the fact, that this device can not work with the TFT35 **V2.**

Baltojikale, 2020-09-30:
Ok my set up is tft35 ve3 v3. Kinda sorted out with sd cloud. So as recomended sd cloud was pluged before power up. thus it consistently connects to wifi and is able to send files. Speed is kinda slow. After prints if everything goes well and if on tft i go back to main screen sd card is once more accesable via network. Though after printing not always sd card is accesable.

oldman4U, 2020-09-30:
@Baltojikale 

You are right. Not to be able to work with the device once your printer grabbed it is a shame and a real show stopper - at least for me. And I do not know if this is because of the software, hardware or a user error. But this ticket is about users which can connect their device to the WIFI network, only in case it is powered via USB, but not without. This is a complete different topic.

Bastlwastl84, 2020-10-30:
i'm having the same Problems with whitespace at name of the SSID and does not connect. if i would change the SSID i need to change a lot of devices

chilman408, 2020-10-31:
> @Baltojikale
>
> You are right. Not to be able to work with the device once your printer grabbed it is a shame and a real show stopper - at least for me. And I do not know if this is because of the [...]

Can we both power the SD Cloud via USB and also have it read as an SD Card?  Will this damage my printer?

oldman4U, 2020-10-31:
Why would you do so?

chilman408, 2020-10-31:
> […](#)

Not sure why the SD Cloud is not powering up in the SD slot, power from micro usb connects via wifi.

oldman4U, 2020-10-31:
Then it is broken and you should contact the dealer you bought it from.

oldman4U, 2020-10-31:
Put the SD Cloud into another device like a card reader or PC and see if it
powers up then.

ant789, 2021-09-26:
Just bought the BTT SD CLoud V1, it does not work in SD slot (PC or ENDER3),
When connecting a 18650 (3.3v) directly to VSS and VDD (on the SD part), it works perfectly.

So, in my case, it seems the only issue is SD port current limitation... But the maximum current draw of the BTT SD is 82 mah (measured at 4.14V), wich doesn't seems too high for SD slot no?

sarumpaet, 2022-04-08:
Here: BTT Cloud blinks (red+green on PCB, blue on ESP) briefly, then turns off when connected to SD port on Geeetech 2004 controller. Works fine if connected via USB.

This is a power issue. I soldered an additional capacitor to the ESP (Vcc/GND), and it gets a bit better (lights stay on, blue light blinks somewhat randomly), but still it doesn't connect to Wifi.

The 2004 apparently supplies only about 3.1 Volts. It appears the MIC5209-3.3YS voltage regulator on it just doesn't cut it even though it's rated for 500mA, and the input capacitor recommended for it is present.

sarumpaet, 2022-04-12:
I spent some quality time with the device, and it seems it's extremely picky concerning **power supply**.

The easiest way to check if power is the issue is **taping off all SD card lines but 3.3V/GND** (for standard SD that's pins 4 and 3). If the device doesn't power on properly then (i.e., doesn't connect to Wifi) - but connecting to Wifi works fine if connected via USB - then power is the issue.

It seems the device is fine with low voltages (testing with a benchtop power supply looked fine down to 3.0V), but the device generates relatively big load spikes (it consumes about 20..250mA), and seems overly sensitive to transient voltage spikes. Even with a 470µF electrolytic capacitor plus parallel 22µF tantalum soldered to the ESP module directly, it would fail to boot with longer power supply lines (coming with about 0.4 ohms of resistance).

I'd not be surprised if there's something weird with the smoothing capacitors within the ESP module.

Easiest and most robust solution is probably to tape off the 3.3V pin of the SD card connector, then powering the device via USB (taping off the 3.3V SD card pin makes sure no current flows from USB to the printer control board).