bigtreetech/BTT-SD-TF-Cloud-V1.0 #4: Can't acces to SD from macOS

BastR, 2020-06-23:
Hi, the module works correctly, I can see its IP through my router, but I can't add the remote drive like on windows... Someone managed to access the folder under macOs?

BastR, 2020-06-23:
Ok it works, IP is 192.168.0.XXX.

In Finder press **⌘ + K**, 
then put your IP like this "**http://192.168.0.XXX/**" (without DavWWWRoot), 
then press **Connect**, 
then agree to connect to an unsecured server,
to finally log in as guest user.

Et voila

closed 2020-06-23 by BastR

oldman4U, 2020-06-24:
Problem reported - problem solved - explained the solution - ticket closed.

We need more users like you;-)

Merci

oldman4U, 2020-09-30:
Hi.

Sorry to bother you, but can you copy any kind of files from Finder to the SD card? I am not able to do so, does not matter what size or format of the SD card. Tried Mojave and Catalina. Only way to get data on it is the direct export using Cura.

You help is very appreciated

Thank you