bigtreetech/BTT-SD-TF-Cloud-V1.0 #5: BTT-SD-Cloud ESP does not see SD card

knive87, 2020-06-28:
Hi friends! Please help me. BTT TF Cloud does not work. Bought on aliexpress BTT SD Cloud. I made all the settings as on video, but ESP does not see the card, or the problem is in the firmware. All connections from ESP to the card rang, everything is OK. But in the terminal it just writes AI Technologies and Ready. Manipulations with the card (replacement with another, formatting in FAT16, FAT32 do not help). Maybe someone came across? You can of course try to re-flash ESP yourself, but for now everything is under warranty.

oldman4U, 2020-06-29:
Hi.

Can you see the green LED on your TF Cloud illuminated, when you insert the SD card?

No idea how it is uploaded, but there is new firmware available since a few hours. Maybe this helps....

knive87, 2020-06-30:
> Hi.
> 
> Can you see the green LED on your TF Cloud illuminated, when you insert the SD card?
> 
> No idea how it is uploaded, but there is new firmware available since a few hours. Maybe this helps....

The green LED is only a short to ground contact located on the SD card slot and it glows when the card shell is pressed. Here the problem is either in the firmware or in the ESP itself. I found it strange that 4.9v from ch340e comes to Rx input instead of 3.3. I think it was a ch340e marriage that damaged ESP and now it gives garbage.

oldman4U, 2020-06-30:
Hi.

Seems you already have an idea where the problem could come from, great. Firmware is unlikely, because this would mean that all other devices shipped so far would have the same issue. Which makes a hardware problem the most likely reason for your problem.

oldman4U, 2020-09-30:
What is the status of the reported issue?

Please let us know

oldman4U, 2020-09-30:
And have you ever tried to connect the SD Cloud to the card reader of a PC? It should mount like an SD card.

Bastlwastl84, 2020-11-17:
> And have you ever tried to connect the SD Cloud to the card reader of a PC? It should mount like an SD card.

The Cardreader and the Printer TFT can access to the SD at the BTT Cloud (read. only) and I can see the files. BUT: Upload is NOT working via Wifi.

I can add the networkadress at "Dieser PC (my Computer)" and than rigthclick at a free place Netzwerkverbindung hinzufügen (add networkconnection)  "http:// "IP Adress""

 i can add (but not upload). 
[image]
[image]

When i add like the PDF  "\\ "IP Adress" \DavWWWRoot" i can not add it
[image]

https://drive.google.com/file/d/18iX8Q9XdyFFONaY8ABgzlLRhHpOZAbaq/view?usp=sharing
https://drive.google.com/file/d/18iEzYULVCYfrQRWGQTRVMoUGilJ0W2TE/view?usp=sharing

xevoss, 2021-06-10:
Hello, I got similiar problem, when inserting SD Cloud module into pc (it's big SD module) it doesn't show up on windows. I've written SETUP.INI to SD card and module connects to wifi and it can be accessed via Network Location so this works well. Can you point me why its not working when plugged as SD card? When plugged in it has red and green Led lit, but it goes off within 10-20 seconds.

madjx, 2022-04-01:
Hello,

I have exatly the problem described. The sd-card is not readable. I can see that there is a volume in the reader, but no TF CARD readable. WIfi is not connected when plugged in SDCARD port.

Thank you for your help.

madjx, 2022-04-01:
#`And have you ever tried to connect the SD Cloud to the card reader of a PC? It should mount like an SD card.`

I can't use BTT SD Cloud has a TF/SD adapter

knive87, 2022-04-02:
> Hello,
> 
> I have exatly the problem described. The sd-card is not readable. I can see that there is a volume in the reader, but no TF CARD readable. WIfi is not connected when plugged in SDCARD port.
> 
> Thank you for your help.

Hello my friend. I apologize for my step, but I will describe my decision in Russian in order to make at least a small contribution to the establishment of relations between the countries.

Собирал проект, описанный на Github по созданию устройства подключения SD карты через Wi-Fi. Параллельно с этим со скидкой заказал готовое устройство, которое при проверке отказывалось работать. Не было подключения по WiFi, карта тоже не обнаруживалась. Продавец с Aliexpress тоже разводил руками. Вся обвязка была один в один копией проекта со своими улучшениями. Было решено залить прошивку, после чего все заработало. Появилось соединение по WiFi, SD карту подключил как сетевой диск.

Вот ссылка на проект и прошивку. Надеюсь решение и вам поможет. Удачи!

Asdf-0123, 2022-08-03:
Hardware error resolution:
https://www.reddit.com/r/SidewinderX1/comments/k4jqt5/wireless_printing_with_a_bigtreetech_tf_cloud_on/