bigtreetech/BTT-SD-TF-Cloud-V1.0 #6: SD Times out & how to update firmware

BigDaddy4, 2020-06-30:
When I first turn on the printer I can transfer files and run a print just fine.
Eventually the card is no longer accessible and I have to shut it completely off and on again to transfer another file. Is there a way to turn off the timeout?

2nd Question is how to update the firmware? I see there is a new file on the Github page.

Thanks
Rick

brutzler, 2020-07-03:
I had same problems, Transferring a file is sometimes a matter of luck. It stays a long time at 99%.
Firmware: They uploaded the bin-file on git a few days ago.

BastR, 2020-07-05:
Hey, guys,

I managed to update the firmware, I proceeded as follows:

-Download NodeMCU-PyFlasher-4.0 here : https://github.com/marcelstoer/nodemcu-pyflasher/releases
-Locate the firmware.bin in your computer
-Connect the BTT SD cloud with USB to the computer and let the app detect the serial port
-Disconnect the BTT SD cloud from usb and reconnect it while pressing the "BOOT" button
-When the app detect the board, release the boot button
-Clic Flash NodeMCU
-When flash is done, unplug et reconnect the BTT SD could to boot in normal mode

Et voilà, I was able to transfer the firmware.bin of my SKR PRO without problems, I also tried to send a 10mo file and the transfer speed is up to 250KB/s

[image]

BigDaddy4, 2020-07-05:
Thank you! Flashing the firmware worked as you stated.

oldman4U, 2020-07-05:
@brutzler Thank you for helping out.

@BastR Could you please be so kind and make a new ticket with the description of the update process, until BTT is able to update the document. Or could you do this?

Thank you

BastR, 2020-07-09:
@oldman4U I'll try to make a PR for this

oldman4U, 2020-07-10:
@BastR 

This would be great. Thank you

SKOREIKO, 2020-07-16:
Did you manage to get this to work on a Mac? @BastR

BastR, 2020-07-16:
@SKOREIKO yes following the steps above

oldman4U, 2020-07-17:
@SKOREIKO 

There is a closed ticket explaining Mac connection.

SKOREIKO, 2020-07-17:
@oldman4U thanks I'll check it out

oldman4U, 2020-07-31:
@BastR 

Could you please help me. I am also working on macOS and got my SD Cloud device. I used TextEdit to generate the SETUP.INI with SSID and PASSWORD in it and converted it to Plain Text before i inserted the SD card into the device. Once powered up, all i got was the green and red LED and the blue one on the back flashing constantly. So i thought i will try to update the firmware, which i believe worked out (attached screenshot). But i still have the problem, that the blue LED flashes, so no connection can be established. Besides our private network i also tried a second WLAN connection (personal hotspot) and removed the SD card, which resulted in a SD read error in the Arduino app. In Arduino all i get is a "Delay".

Thank you for your help

[image]

oldman4U, 2020-07-31:
@BastR

After the next reset/restart or what ever it worked.....

THANK YOU

chilman408, 2020-09-15:
> Hey, guys,
>
> I managed to update the firmware, I proceeded as follows:
>
> -Download NodeMCU-PyFlasher-4.0 here : https://github.com/marcelstoer/nodemcu-pyflasher/releases
> -Locate the firmware.bin in [...]

Thanks for the instructions.   I was never able to flash app to detect the BTT SD Cloud but I used Arduino to detect the com port on Windows. 

Since the app could never detect the board, I couldn't hold the boot buttom until something is detected.  I just held the boot button for 30s and flash the firmware to the port detected by Arduino.  The screen that resulted look exactly the same as your screenshot.

sstefanov, 2023-10-23:
> Hey, guys,
>
> I managed to update the firmware, I proceeded as follows:
>
> -Download NodeMCU-PyFlasher-4.0 here : https://github.com/marcelstoer/nodemcu-pyflasher/releases -Locate the firmware.bin in [...]

Please, paste here full command line.

edugfernandez, 2023-11-16:
Hola:
La diferencia al actualizar en que se cambia??
La acabo de comprar