bigtreetech/BTT-SD-TF-Cloud-V1.0 #8: Do you plan to open the source code?

kasedy, 2020-07-10:
I'm working on my DIY project of extending ender-3 printer with WiFi capability. I like your product. It can be used as remedy of fast gcode transfer from octoprint to printer's sd card. But I still miss some functions. I would like to improve security and make it compatible with octoprint (through plugins). Are you planing to open the source code to the community? I think it will beneficial for the product as customers like me will be able to extend it and use in their DIY projects.

maketo3D, 2020-07-18:
I think the same. If we could have the scheme to be able to take out some serial pins to connect to the board, ESP3D could be installed and thus have a mini-octoprint with fast sd access.

drdelaney, 2020-07-29:
My best guess is that this is built around a mix of ardyesp/ESPWebDAV, FYSETC/ESPWebDAV, and some custom code changes.

Since it looks like the official ESPWebDAV reads the ssid/password from the arduino sketch, and the fysetc one looks to read from the setup.ini file and store said parameters in EEPROM.

Due to some of the more recent changes to the fysetc code, I am going to guess based on that more than its original source from ardyesp.

brutzler, 2020-07-31:
Try this:
[https://cloud.mail.ru/public/3zqi/3ecDR82hg](url)

maketo3D, 2020-08-05:
> Try this:
> [https://cloud.mail.ru/public/3zqi/3ecDR82hg](url)

Wrong link? there is nothing

brutzler, 2020-08-06:
sorry, try the link again.

sebi5361, 2020-09-11:
@brutzler: Thanks for the link!
I didn't dive into the code to check if some pins are moved, but I guess that this hardware's wiring is similar to the one depicted in ESPWebDAV.
Do you confirm?

AlbrechtL, 2021-02-13:
Here you can find an alternative firmware for the devices.
https://github.com/AlbrechtL/BTT_TF_Cloud_AFW

I created it because I needed some custom features. Because the original source code is not available I put together public ESP source code to create this alternative firmware.