bigtreetech/BTT-SD-TF-Cloud-V1.0 #9: Impossible to connect

hantrax, 2020-07-12:
HI..I have received yesterday a BTT SD Cloud v1.0
I have followed your instruction and I have connected it to my wifi.
I have connected it to my BTT TFT 35 and I ran a print. after a few minutes the print stopped.
I have downloaded new firmware and flashed it to the BTT SD Cloud
Now I can't connect it to my wifi network.
any idea?

NAPCAL, 2020-08-28:
Just found this out on my BTT Pro V1.1 using Marlin 2.0.6.
Configuration_adv.h 
When selecting #define SDCARD_CONNECTION ONBOARD (Configuration_adv.h), the TF cloud device will not start the Blue led blinking (when trying to connect to WIFI) and as a result, can not connect to the device via computer for file transfer.

By changing this selection to #define SDCARD_CONNECTION LCD (Configuration_adv.h) will allow the TF cloud to connect for usage.

oldman4U, 2020-09-27:
Hi hantrax.

What is the status of the reported issue?

oldman4U, 2020-09-30:
Please help the community and close the ticket once you do not need it anymore. Thank you

hantrax, 2020-09-30:
After a firmware update I can use it..but the link is not stable...

closed 2020-09-30 by hantrax

oldman4U, 2020-09-30:
Not stable in terms of...?