bigtreetech/BTT-SD-TF-Cloud-V1.0 #10: Network connect on Linux

gjeff78, 2020-07-14:
Hi,
Witch protocol is used to connect once setup is done ?
It works nicely but I would like to use the connexion with my Linux machine.

Thanks

BastR, 2020-07-15:
It use Webdav

gjeff78, 2020-07-16:
Thanks, I'll try to use it

gjeff78, 2020-07-18:
Hi,
On linux, Webdav connexion needs a username and password, is there default one for the firmware ?

Woit, 2020-08-25:
@gjeff78 after several attpempts i connected with user "admin" and empty password

iondulgheru, 2020-10-09:
@Woit , I tried connecting to http://admin@put.your.ip.here/DavWWWRoot, it seems to communicate with the device, but the response is this:
```Not found
URI: /DavWWWRoot Method: GET```

feiticeir0, 2021-01-09:
It's not working on Linux using CIFS. 
If not specified, it uses root or whatever user is trying to connect. 
If specified in /etc/fstab with user=,password= 
it gives the following error:
mount error(111): could not connect to <IP_ADDRESS>Unable to find suitable address.
(the lack of space after the IP address is not a type.. It does not have one)

feiticeir0, 2021-01-09:
> @gjeff78 after several attpempts i connected with user "admin" and empty password
> 
> And also on my Mac OS it requres connetction string `http://admin@put.your.ip.here` instead of `http://admin@put.your.ip.here/DavWWWRoot`

I've tried with admin and still not working...

feiticeir0, 2021-01-09:
Even with webdav, using Thunar, it's not working... connection refused..
davs://admin@<ip_address> (with or without DavWWWRoot)

Leo-PL, 2021-01-27:
With KNetAttach, it worked for me, when I used folder name "/" instead of DavWWWRoot.

feiticeir0, 2021-01-28:
> With KNetAttach, it worked for me, when I used folder name "/" instead of DavWWWRoot.

How did you do it ? 
I'm still not having sucess with it and BTT still is doing nothing..

Leo-PL, 2021-02-05:
@feiticeir0 please see here: 
[image]
WiFi-SD-Card-3D-Printer is the hostname which ESP-SD requests via DHCP from my router, make sure it's accessible under this address, or use IP directly. I use OpenWrt, so no explicit configuration was needed for my router to resolve hostname.

feiticeir0, 2021-02-08:
@Leo-PL Thank you. I don't use KDE, but did try your settings in Thunar and even in the console and no luck... Don' know if I'm missing some configuration or some package support, but no luck...

froqstar, 2021-02-26:
@feiticeir0 with nautilus/gnome, it works for me via `dav://<IP>/`, no username or password required.

feiticeir0, 2021-03-17:
Hi all !
Since I now use a Mac, I was able to connect to BTT Cloud , but only using admin/admin has username and password .

sarumpaet, 2022-03-14:
The WebDAV implementation on the device seems to be a bit weird.

- Connecting to it using a browser gives "not found" for me (though http://192.168.x.x/SETUP.INI _does_ return the SETUP.INI file). It would probably be nicer if the device would generate normal HTML listings.
- Using davfs2 to connect to it fails with `the server does not support WebDAV`
- Using the [cadaver](http://www.webdav.org/cadaver/) command line client works just fine.
```
$ cadaver 192.168.x.x
dav:/> ls
Listing collection `/': succeeded.
        SETUP.INI
```

Note the `DavWWWRoot` thing in the docs is just a Windows hack. Apparently Windows needs this to trigger the WebDAV protocol (instead of SMB) while at the same time opening the root dir. So on Linux/Mac, just leave this away.

AlexYzhov, 2022-12-07:
> Note the `DavWWWRoot` thing in the docs is just a Windows hack. Apparently Windows needs this to trigger the WebDAV protocol (instead of SMB) while at the same time opening the root dir. So on Linux/Mac, just leave this away.

You are right.

MacOS finder access http://admin@<IP> with empty password simply works.