bigtreetech/BTT-SD-TF-Cloud-V1.0 #12: IP Visible but not connecting

iankds05, 2020-07-21:
Have received the SD Cloud SD module but despite making the setup.ini file and the module connecting to my network, I cannot connect to it from any computer. It has obtained an IP address from the DHP server but other than being pinged successfully there is no way I can make a network connection from any of my network devices (Computer/Phone/Android Box). I have flashed the firmware BIN file but is has not resulted in any successful connection. Grateful for any ideas

oldman4U, 2020-07-23:
You have to mount it on your computer.

Have you tried http://*your IP address*?

iankds05, 2020-07-23:
It has been powered both from USB connected to the PC (This is how it got flashed with firmware) and also from the SD card slot on the 3D printer. In all cases it obtains a local IP address, and yes I have tried to access just the //IP address, both with just the IP and also //IP address//DavWWWRoot.  Neither option works as windows says the IP exists but the file system is not obtainable.

oldman4U, 2020-07-24:
http://ip

oldman4U, 2020-07-25:
Any progress?

iankds05, 2020-07-25:
No afraid not. Tried all sorts of things but no luck at this point. Guess its just a bad unit and only fit for the trash bin. The design has been copied from a GitHub design with a few modifications, so I am going to make my own from the original  WebDev design.

oldman4U, 2020-07-26:
You have to use http:// and your ip address. Just the ip address will not work. But if you can make your own webdev device you know this already. If it is broken you should be able to get a replacement.

oldman4U, 2020-08-03:
Hi iankds05

Please close this ticket in case you do not need it anymore.

Thank you

iankds05, 2020-08-04:
No good result with this issue so assumed item is faulty

iankds05, 2020-08-04:
If anyone has any further developments with this product I'm prepared to re-open else assume it is defunct

closed 2020-08-04 by iankds05

oldman4U, 2020-08-04:
Have you ever tried 

Http://youripaddress

And what is your SSID

There is no need to close a n actively used ticket but it makes no sense to keep one opened but not answering other users which try to help.