bigtreetech/BTT-SD-TF-Cloud-V1.0 #13: Wifi network name with whitespaces.

aluqueFIBAO, 2020-07-27:
I just received the sd card and I can't get it to connect to my home wifi network, I've been trying and I think it's because of the blank spaces on the SSID, Red Piso 2G. 
I have tried using quotes but with no result. Using another temporary network without spaces it connects without problem. The password contains characters like # and $, although I think this is not a problem.
I appreciate any help.

brutzler, 2020-07-29:
You could check the length of your ssid and/or pw.
It is limited to 50 characters.

oldman4U, 2020-07-29:
Hi brutzler.

Thank you for helping out. I got my SD Cloud today, but your input is more than very welcome!

aluqueFIBAO, 2020-07-29:
The ssid and the password are less than 50 characters. I tried it with the same network deleting the whitespaces and it works. Is there a way to use a ssid with whitespaces? I don't want to change the name of the wifi network.

drdelaney, 2020-07-29:
Looking at the fysetc firmware source (btt, why do we not have any source?!?!?!?!), they are stripping all spaces from the SSID, changing them to a null entry.
https://github.com/FYSETC/ESPWebDAV/blob/1ac7da52600b9c6b2089bbe9e8c54ebf9848a197/config.cpp#L40

brutzler, 2020-07-31:
Yes, they do the same:

`    // Delete CR and Space.
    if (ch == '\r' || ch == ' ') {
      continue;
    }`

NAPCAL, 2020-08-28:
It will not connect to an SSID that has space(s) in the name, I had to remove the spaces from mine to get it to connect.

Also if you have issues connecting to the device from your computer.
Just found this out on my BTT Pro V1.1 using Marlin 2.0.6.
Configuration_adv.h 
When selecting #define SDCARD_CONNECTION ONBOARD (Configuration_adv.h), the TF cloud device will not start the Blue led blinking (when trying to connect to WIFI) and as a result, can not connect to the device via computer for file transfer.

By changing this selection to #define SDCARD_CONNECTION LCD (Configuration_adv.h) will allow the TF cloud to connect for usage.

pat92fr, 2020-10-17:
Hello. Support of whitespaces in SSID and PWD will be added in future release ?

brutzler, 2020-11-14:
IMHO there will be now future release from BTT. They do not develop any software for this module. only copying code stuff from internet.
But you can do it for your own.
Look two posts above. There is the matching part of code. And a link to the sources. Try it.

mhegering, 2021-02-25:
Has anyone a modified version which can read SSIDs with whitespaces? Im not that into coding this Wifi thing, so it would be nice if you could link a respository. @brutzler did you figured it out?

CliveDent, 2021-03-12:
Removing 

` || ch == ' '`

should be enough.

RyanEwen, 2021-05-17:
I am looking to update my board with support for spaces in the SSID. Has anyone actually done this?

Arduino is new to me. I installed the s/w, added the URL for ESP8266 boards, set the board as `Generic ESP8266 Module`, and tried to compile. I am missing the ESPWebDav library, though. And I am worried if I don't have the correct board settings that I may brick the board.

yzheka, 2023-05-19:
> Yes, they do the same:
> 
> ` // Delete CR and Space.` ` if (ch == '\r' || ch == ' ') { continue; }`
> 
> And IMHO BTT is not using any sources. They only copied the binary file from here: https://cloud.mail.ru/public/3zqi/3ecDR82hg
> 
> I checked it on byte-level. It is the same as I got from BTT-Support. At this link you will find sources too. With this I was able to modify as needed.
> 
> Happy coding!

can you please advice me what version of ESPWebDav library should I use. I tried 2.0.0 - 2.1.5 and no one worked. Compiler throws me errors like `'class ESPWebDAV' has no member named 'init'` and `'class ESPWebDAV' has no member named 'rejectClient'` and `'class ESPWebDAV' has no member named 'rejectClient'`