bigtreetech/BTT-SD-TF-Cloud-V1.0 #14: Reading failed! exception 28(...

fbforos, 2020-08-17:
Dear colleagues,

when I try to retrieve the ip number I received this message from the device. I have tried differents TF cards and always the same error.

Please will you have any Idea about what could be happening?

Reading failed!
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  00000000 feefeffe 3fff0554 4020233e  
3fff1560:  feefeffe feefeffe feefeffe 40208569  
3fff1570:  3fffdad0 00000000 3fff0554 40209074  
3fff1580:  feefeffe feefeffe 3fff0570 40100a28  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
115200

NAPCAL, 2020-08-28:
If your SSID has space(s) in the name it will not connect. Also, the device will not work with SD card interfaces that use SIO instead of SPI.

oldman4U, 2020-09-27:
Hi.

What is the status of the reported issue?

fbforos, 2020-09-29:
oldmand4U, after your interest about the status I've tried today again.
I´ve discovered that the problem is the length of the Wi-Fi password. It not meets with the standard of the WPA2-AES.
Which is the longest password that can be used with this system? I can discover but it will take a lot of time and it´s better ask :-).
The password that I was using was of 48 characters. May anybody modify the program to meets with the standard.?
Thanks

oldman4U, 2020-09-30:
Please read ticket #1 which is related to this

oldman4U, 2020-09-30:
https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/issues/1

fbforos, 2020-09-30:
Thanks. I will read this afternoon.

Enviado de Samsung Mobile

Obtener Outlook para Android

oldman4U, 2020-09-30:
Please do so and let us know. And no white spaces in the SSID.

fbforos, 2020-10-02:
Oldman4U as I told in my post at this moment the system is working with a password shorter than 48 characters, so the max lenght is shorter than 48. May be that I try to review the soft but at this moment I'm working in a diferent project.
Thanks for your help and interest

oldman4U, 2020-10-02:
Great to hear that your problem is solved. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-10-02 by fbforos

GhostbustThat, 2022-01-11:
I'm also getting this error, but I don't believe the issue is my SSID or password. No spaces and few enough characters for ASCII. Purchased the BTT unit with a TF card provided, tried another TF card just in case and got the same result. Error is as follows:

`open file failed
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  feefeffe feefeffe 3fff0554 4020233e  
3fff1560:  feefeffe feefeffe feefeffe 40208569  
3fff1570:  3fffdad0 00000000 3fff0554 40209074  
3fff1580:  feefeffe feefeffe 3fff0570 40100a28  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
115200`