bigtreetech/BTT-SD-TF-Cloud-V1.0 #17: Only connects for a few seconds when starting up...that's it

chilman408, 2020-09-06:
It loses connection 30 second after... this is useless.

oldman4U, 2020-09-27:
Hi.

You mean WIFI connection or that you can not see the content of the SD card?

Have you tried to connect only USB power?

chilman408, 2020-09-27:
Yes, USB power and card side not connect to reader, wifi works fine.

oldman4U, 2020-09-28:
I am still not sure what you mean, sorry.

Please let me know more about your configuration, mainboard, TFT and firmware, as well as how they are connected.

Or do you only test with USB, then I would like to know if you have done the firmware update of the WIFI module.

chilman408, 2020-09-28:
I only tested with usb on my PC to check Wifi.  That worked.

Then I plugged in the card as an SD card to the printer.  That only connects for a few seconds so I could see the content of the card via wifi but I cannot finish uploading any file.

This happened before and after upgrading the Firmware (so yes, I did updated the firmware per instructions on the other "issue" post here)

oldman4U, 2020-09-28:
Which mainboard and which TFT?

When you have the device in the TFT and the mainboard connected and you turn it on, what does the blue LED during the first 2 minutes?

chilman408, 2020-09-28:
Creality V2.2 and I have used both the BTT TFT35 E3 V3 and the stock LCD12864 (same issue with both).

I have the check the LED situation when I get home, but from recollection, sometimes the Blue LED blinks and sometimes it is solid Blue, other time no Blue LED.

oldman4U, 2020-09-28:
Let me know exactly what it does within the first two minutes after power up.

Thank you

chilman408, 2020-09-28:
Ok, no problem.

chilman408, 2020-09-30:
> Any news?

Sorry about the late reply.

Immediately after turning on the printer, the blue led is off for 15 seconds, then it blinks about 2x per second for 90 seconds, then it stops blinking.

oldman4U, 2020-10-01:
How many digits is your password?

Above you wrote it connects to WIFI. Is this still the case or what have you to do to make it work and does it blink different then?

chilman408, 2020-10-01:
> How many digits is your password?
> 
> Above you wrote it connects to WIFI. Is this still the case or what have you to do to make it work and does it blink different then?

Password is 16 digits

It connects to wifi on USB connection only or I guess USB power.  Not sure about how it blinks then.

oldman4U, 2020-10-01:
My experience is, that when the blue LED is off for 15-20sec after startup and it flashes then (on my system 41x) it has connected to the WIFI once it stops blinking.

Otherwise it flashes every 20-30sec one or two times - which means it tries to connect.

Check how it flashes when you are able to connect to it.

effgarces, 2021-01-02:
While  I don't have this issue with the TF Cloud, I've noticed something that can, maybe, provide a clue. My TF Cloud works reasonably well in conjunction with a TFT24 and a SKR 1.3, but the bug with SSID's with spaces is annoying,  so I've decided to try using the ESPWebDav code from Fystec https://github.com/FYSETC/ESPWebDAV, this version already has this fixed. To my surprise this bug manifested itself when using this firmware. Using it, my TF clouds connects to wifi if I  use it connected to a usb port, but as soon as I try using it connected to the SD it no longer connects. Reverting to the TF-Cloud original firmware, provided here in this repo, I started connecting again.
As a side note, if anyone wants to use the ESPWebDav code, above  you will need ESP Core 2.4.0 and SDFat 1.0.5.
When I get enough free time I'm planning to experiment with the code over at https://github.com/d-a-v/ESPWebDAV, that seems to be better maintained.

oldman4U, 2021-01-02:
Hi. Sounds great. Is there a chance to make this work using macOS?

effgarces, 2021-01-02:
Honestly, I don't know and I don't have OSX to test, but after checking out almost all forks, to me, it seems that the d-a-v  version is the most likely candidate to work, the downside is that there's no ini setup support, so you have to compile a version for your own wifi network. If it works, maybe we can hack together something to provide that funcionality.

RobThree, 2025-01-10:
Connection is utter sh*t:

[image]

And it's most DEFINITELY this piece of crap. I have _AT LEAST_ 15 Wemos D1 mini's (also ESP8266'es) around the house reporting temperatures and all kinds of other stuff and they ALL have been working fine for the last 7 years: 

[image]
(These are just the temperature ones, I have others around the house, and only the last 30 days but I can go back all 7 years and none of them have nowhere near the connection loss as this thing).

I use these things SO much that I have [literally 20 on stock](https://github.com/user-attachments/assets/a186bc3a-27a9-4f3b-a6d5-71f6679724e9) and I have no problem getting any of these online from the exact same distance at the exact same location. And this piece of crap can't handle a stable WiFi connection for even a minute. I'm looking at it currently, 1 meter(!) away from a decent WiFi access point with actual line of sight. And, no, further or closer doesn't help anything.

Yes, I am certain it is the correct IP, I have been able to contact it -briefly- at times. I even flashed [this alternative firmware](https://github.com/AlbrechtL/BTT_TF_Cloud_AFW) on it but that didn't help either.

I am considering taking the ESP off and trying an ESP from a Wemos D1 mini on it. But I don't think it'll help. I'm guessing it's either the ESP-12S that's on this piece of junk OR a lack of some local capacitance or something that the Wemos boards DO have and this thing doesn't.