bigtreetech/BTT-SD-TF-Cloud-V1.0 #18: What it does and what not - my 2 cents

oldman4U, 2020-10-01:
Hi.

For a few days I have been testing the BTT SD Cloud v1.0 (firmware June 29th) device now using the following hardware:

128MB and 8GD SD Card, formatted in FAT16 or FAT32 with the SETUP.INI (with SSID and Password) in a PLAIN TEXT text file on it all the time.

macOS 10.14. and 10.15, no Windows, no Linux.

TFT28 v1.1 and TFT35 E3 v3 latest firmware from Sept. 26th
SKR E3 DIP v1.0 and SKR Pro v1.1 firmware 2.0.7

I have been testing the things listed below using the SD Cloud device connected to USB only, and installed in the two TFT's. I bought the SD Cloud and not the TF Cloud, so there is no way to use it attached to the SKR's.

To mount the SD Cloud, I used: http://IP Address, even the handbook describes a different method, which - at least on the Mac - does not work.

Here is what works:
I was able to -
*connect using any kind of user name (and Password) or Guest
*unmount it and mount it again
*read the already existing files and copy them from the SD Cloud onto the Mac
*save a (BigTreeTech) gcode file directly from Cura to the SD Cloud
*print from the SD Cloud
Thats it. I mean this is exactly what has been advertised and why I bought it.

But I hoped for more, so here is my list of what worked unreliable:
*copy files (gcode, firmware,..)from macOS to the SD Cloud via WIFI connection
*rename existing files
*connect from several Mac's and get all updated in case something changes

And this did not work at all, but I believe it should - for a useful workflow:
*Ability to access the SD card of the SD Cloud after a print has finished
*Ability to change the SD card without the need to reboot the printer

I can not reproduce the issues some users reported, where the device does only connect in case it is connected to USB, or where it fails after some time.
My device has the green and red LED illuminated and the blue LED flashing fast 2-3x, once it gets power. After that, there is not activity for app. 20sec, followed by 41 times the blue LED blinking. After that I can connect via WIFI. In case it can not find the SETUP.INI on the SD card, or it can not find the WIFI network, the blue LED will blink 2-3x every 20-25sec. In case I resolved the problem I the meantime, the blue LED with blink 41x and connect to the WIFI.

Especially the fact that you have to restart the printer, to be able to load new files after a print, and that it is not possible to copy files from the computer to the SD Card of the Cloud device, limits the functionality in a way, that I would not buy it again.

If someone has a solution to one or several of the issues above, please let me know. I will update this ticket whenever needed.

Thank you and happy printing

oldman4U, 2020-10-09:
Hi.

I have been using cmd-K to connect to a server and then http://IPaddress.

Hope this helps

Foxcor, 2020-12-04:
@iondulgheru 
@oldman4U 
http://agriculture.montana.edu/it/support/smb-macs.html

NGavanozov, 2021-01-24:
For me, Ubuntu 20.04 connects like this:
Connection to server, protocol:
dav: //<ip_address>

ipmcc, 2024-07-21:
I have googled around, and read the manual, etc. I can connect via http, but if I plug the BTT SD Cloud v1.0 into an SD reader (or my machine's built in SD card reader), it does not register with my Mac as an SD card volume. In fact, nothing happens at all. If I take the µSD card (without the BTT SD Cloud v1.0) from my printer (or from the BTT SD Cloud v1.0) and plug it into my Mac with a reader, it works fine. 

At this point, I'd say I'm just very confused about how this thing works, or is supposed to work. Do the printers not mount this as a drive but access it via DAV? Is that universal? (FWIW: I have a BambuLab P1S)

My expectation when I bought this was that it would act like a normal, mountable SD card (while not re-writing via WiFi). It seems like it acts like some 'special' SD Card. I also can't tell if the communication with my Mac is happening over the USB connection vs over WiFi vs over the SD Card interface.

I don't mean to be cranky, but I feel like this device is not performing as advertised.