bigtreetech/BTT-SD-TF-Cloud-V1.0 #19: I didnt crate ip adress

Mecitinceoglu, 2020-10-17:
Hello to everyone ,
I'm trying to get ip address from Arduino IDE but it always says delay.

It doesn't give any i p. I have tried with different sd cards. But the problem is the same. Is there something I did wrong?
Thank you.

Mecitinceoglu, 2020-10-17:
[image]

oldman4U, 2020-10-18:
Have you made the firmware update?

Mecitinceoglu, 2020-10-18:
> Have you made the firmware update?

Hi again ,

Yes ı update to firimware.I take firmware from this page and upgrade with bin file.

oldman4U, 2020-10-18:
So you have an SD card in the unit which has the SSID and Password of your network and there are no white spaces in your SSID? The SD card is <8GB and FAT32 and the textile withe the SSID/Password is plain text?

Mecitinceoglu, 2020-10-19:
Yes I Have a 16 gb Sd card and there arent white space in my ssid an password SSID/Password is plain text.

Just I use 16GB SD card is it problem ?

Mecitinceoglu, 2020-10-19:
I tried with 4 gb sd card i'm live in same problem

oldman4U, 2020-10-19:
Your router has DHCP activated?

Maybe your router shows new device and the IP address?

When you start, what does the blued LED during the first 2 minutes?

Mecitinceoglu, 2020-10-22:
Hi ,

i still can't get ip.I have a router but btt cloud not use router.It must produce ip.

Blue LED is not lit at startup.It starts to flash after a while, but no ip appears on the serial port screen.

And only Delay is written on the screen

oldman4U, 2020-10-22:
The btt must get the ip from the router.

How long does it flash and does it stop for at least one minute?

Mecitinceoglu, 2020-10-22:
Thank you I solved the problem :)
I thought that will create ip adress BTT :)
I should write to my self ssid and password.
Thank you for support.

oldman4U, 2020-10-22:
Perfect!

Please close the ticket.

Thank you

Mecitinceoglu, 2020-10-22:
Thank you again.

closed 2020-10-22 by Mecitinceoglu

oldman4U, 2020-10-22:
👍🏻

Have fun