bigtreetech/BTT-SD-TF-Cloud-V1.0 #21: Issue after connecting to WiFi

Bastlwastl84, 2020-11-07:
after connecting:
after i changed the WiFi-Name without a whitespace and updated the Firmeware before update it was not possible that i could connect to wifi. now i got at browser:
Not found
URI: / Method: GET

and at \\"IP-Adress"\DavWWWRoot the error that it can not connect.

oldman4U, 2020-11-07:
Servus.

Not sure I understand your issue right. Where do you get this issue after doing what. I understand right that you have been able to connect the device to your WLAN and now you do what?

oldman4U, 2020-11-07:
The DavWWWRoot method hasn't been working for a single user so far.

Which OS are you using to connect to the device?

Bastlwastl84, 2020-11-07:
Using win10 64bit 20h2
[image]
[image]

oldman4U, 2020-11-08:
Don't know how WebDAV servers can be mounted using Windows but on macOS I used http://ipaddress/

Bastlwastl84, 2020-11-11:
> Any news?

nope. not working  LED is blinking 2-3x short while click on the networkadress, but only an error cames:
[image]

[image]

oldman4U, 2020-11-11:
So you have been able to access the SD card, because you can see the file, right?

Have you tried a shorter file name?

How have you verified the IP address?

Bastlwastl84, 2020-11-11:
> So you have been able to access the SD card, because you can see the file, right?
> 
> Have you tried a shorter file name?
> 
> How have you verified the IP address?

same happens with 8+3, IP is correct (looked at Router)

oldman4U, 2020-11-12:
Don't know if it is the same on Windows but on mac you get a network volume. Reading my 2 cents ticket you will see that I can only add files before print and have to restart the printer after print for uploading new files. Copying from finder is unreliable and so I have to load the file directly from Cura. This works and printing. Nothing else.

Bastlwastl84, 2020-11-12:
> Don't know if it is the same on Windows but on mac you get a network volume. Reading my 2 cents ticket you will see that I can only add files before print and have to restart the printer after print [...]

i tried to upload luc-github ESP3D and it starts, but can not connect to the SD-Card.
if i try with arduino the fystec sd cloud firmware i got a lot of "was not declared in this scope" after clicking at compile
i found some ftp git, but none with connecting to router
i found some webdav - git, but was not able to compile (i'm a beginner at this)

oldman4U, 2020-11-12:
Please send s picture of the hardware.

oldman4U, 2020-11-13:
Picture?

Bastlwastl84, 2020-11-13:
[image]
[image]

oldman4U, 2020-11-13:
Ok. Forget about the ESP from luc, this is for the WIFI module.

Bastlwastl84, 2020-11-14:
> Ok. Forget about the ESP from luc, this is for the WIFI module.

luc replied: https://github.com/luc-github/ESP3D/issues/454

oldman4U, 2020-11-14:
Yes. Like I wrote.

Bastlwastl84, 2020-11-16:
> Any news?

nope. din't tested it again and was waiting if anybody has an other solution like FTP, SMB, Websurface or another WebDAV solution that will work

oldman4U, 2020-11-16:
Ok. Sorry that I can not help with Windows.

oldman4U, 2020-11-16:
I expect Google knows how to do it:

How do I set up WebDAV for Windows 10 or 8?
Add a network location. Right-click on This PC. ...
Select Next. ...
Choose a custom network location. ...
Enter the WebDAV URL. ... http://ipaddress/
Enter your User Name and Password. ...
Name the network location (Optional) and select Next. ...
Select Finish.
You can now drag files and folders to and from your computer and the site's folder.

Bastlwastl84, 2020-11-17:
> I expect Google knows how to do it:
>
> How do I set up WebDAV for Windows 10 or 8?
> Add a network location. Right-click on This PC. ...
> Select Next. ...
> Choose a custom network location. ...
> Enter the [...]

The Cardreader and the Printer TFT can access to the SD at the BTT Cloud (read. only) and I can see the files. BUT: Upload is NOT working via Wifi.

I can add the networkadress at "Dieser PC (my Computer)" and than rigthclick at a free place Netzwerkverbindung hinzufügen (add networkconnection)  "http:// "IP Adress""

 i can add (but not upload). 
[image]
[image]

When i add like the PDF  "\\ "IP Adress" \DavWWWRoot" i can not add it
[image]

https://drive.google.com/file/d/18iX8Q9XdyFFONaY8ABgzlLRhHpOZAbaq/view?usp=sharing
https://drive.google.com/file/d/18iEzYULVCYfrQRWGQTRVMoUGilJ0W2TE/view?usp=sharing

oldman4U, 2020-11-19:
Hi.

When you read my 2 cents ticket you will see, that I am also not able to upload a file directly from macOS Finder, but I have to export it directly from Cura to the Cloud - this works great. I thought this is a macOS related issue, but seems to be also a problem using Windows;-(

Bastlwastl84, 2020-11-21:
> Hi.
> 
> When you read my 2 cents ticket you will see, that I am also not able to upload a file directly from macOS Finder, but I have to export it directly from Cura to the Cloud - this works great. I thought this is a macOS related issue, but seems to be also a problem using Windows;-(

not working by me with win10 pro 64 20h2

oldman4U, 2020-11-21:
Try to save directly the the cloud device from Cura.

Bastlwastl84, 2020-11-22:
> 
> […](#)

comes an error-Message

oldman4U, 2020-11-22:
Ok. I am out of ideas. Sorry

Bastlwastl84, 2020-11-27:
> Ok. I am out of ideas. Sorry

so, i'm back from hospital, now i had tested it again and with no success . Other solutions would be FTP, but i can't coding software. if someone can programming a small firmware? or adjusting luc-github his ESP3D 3.0 Alpha Firmware that it can Upload to the SD-Card. It has already FTP onboard. the firmware can run on the stick, but can not connect to the MicoSD-Card.

oldman4U, 2020-11-27:
There is a WebDav server running on the Cloud Device and I am not aware that there is another method available to connect. For me there are two possible reasons.

1. there is still a problem with the setup or the SD card

2. the hardware is broken.

First I can not help further because I can not help with Windows or fixing the card and second your dealer has to replace it.

Sorry

Bastlwastl84, 2020-11-27:
> There is a WebDav server running on the Cloud Device and I am not aware that there is another method available to connect. For me there are two possible reasons.
>
>     1. there is still a problem [...]

it can read the SD (cause else it wouldn't connect to the WiFi-Network and i couldn't see the IP at Router) and i can see the Files at Printer what are at the MicroSD on the SD-Cloud, if i would copy (put out of SD-Cloud and put it in a Cardreader and than back) files on Card. But not connecting/Uploading from Windows possible. Tested with 2 PCs. Tested with Powered from TFT, Cardreaders, USB. Luc Githubs Firmware i can Upload, but this solution can not connect to the SD-Card. everytime the same error messages. 

Cause of this i had asked of any other solution if there would be a ftp-project or websurface would possible.

oldman4U, 2020-11-27:
I understand.

Like I wrote before . I am not aware of any other solution. Maybe someone else knows another solution.

Foxcor, 2020-12-04:
@Bastlwastl84 
SD Cloud only works with its own firmware
https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/blob/master/Firmware/Wifi_SD.ino.generic.bin
> Luc Githubs Firmware i can Upload
This is for BTT ESP-01S module!!!
Windows uses the SMB protocol. SD Cloud connects as a network drive.
[image]
[image]

Bastlwastl84, 2020-12-04:
> @Bastlwastl84
> SD Cloud only works with its own firmware
> https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/blob/master/Firmware/Wifi_SD.ino.generic.bin

same i did. Blue LED goes on, when i try to connect, LED goes off and at PC the error Message came.

Foxcor, 2020-12-04:
@Bastlwastl84 
Full format TF card and create new SETUP.INI

Bastlwastl84, 2020-12-04:
> @Bastlwastl84
> Full format TF card and create new SETUP.INI

i allso tried 128MB, 8GB, 16GB and 32GB MicroSD-Cards

Bastlwastl84, 2020-12-04:
i also tried this: https://support.netdocuments.com/hc/en-us/articles/205212850-WebDAV-as-a-Mapped-Drive

oldman4U, 2020-12-05:
Once the blue led goes off, what happens with it the next two minutes?

Bastlwastl84, 2020-12-05:
It keeps off till i try to connect, than it goes on for some seconds. At my Router i can see, that it had connected. I tried with powered by USB and tried powered by Cardreader. Had for 30 Seconds sucess to connect and made an errormessage after i tried ro upload a file 6.3 length. Renamend before Testupload to winkel.gco. after that error i couldn't connect no more an explorer was freezing while trying. Von meinem Sprechknochen gesendet.
-------- Ursprüngliche Nachricht --------Von: oldman4U <notifications@github.com> Datum: 05.12.20  11:12  (GMT+01:00) An: "bigtreetech/BTT-SD-TF-Cloud-V1.0" <BTT-SD-TF-Cloud-V1.0@noreply.github.com> Cc: Bastlwastl84 <thomas.aumeier@gmail.com>, Mention <mention@noreply.github.com> Betreff: Re: [bigtreetech/BTT-SD-TF-Cloud-V1.0] Issue after connecting to WiFi (#21) 
Once the blue led goes off, what happens with it the next two minutes?

—

oldman4U, 2020-12-06:
Yes, you experience more or less the same at least many others have been reporting, including myself. Access to the device is unreliable.

I found mine and tested also smb, like recommended in the other ticket. At least from macOS, this does not work. All that works for me is http:// and saving directly from Cura onto the cloud device. Even copying from Finder level does not work - which ends in an error message and a disconnected device (2x flash of the blue LED every 20sec.

Here on my MacBook I even have to remove the SD card and push it into the Cloud device, to make it connecting to the WIFI.

Wasted time and money

Bastlwastl84, 2020-12-07:
it's better to invest the money in a Pizza. 
Perhaps can anyone create a code for FTP upload with an asyncftp or anything like that as upload option.
i had seen there are a lot of FTP codes on github, but i can't write a code.

Adinfo78, 2020-12-12:
esterday, two ordered WiFI modules came, one SD, the second TF, in the evening I could not get to the SD card connected to the module - I got a connection to the WiFi network, but I did not enter the SD card.

At 1:50 am I woke up and it occurred to me why WebDav didn't start - in newer Windows 10 builds the SMBv1 protocol is disabled by default.

A moment ago, I installed / enabled the SMBv1 protocol in my Windows 10 computer - the connection to the card in the WiFi module works right away!

After enabling SMBv1 Now creating a shortcut to the WiFi module after the IP address and the folder DavWWWRoot with no error message and the shortcut \\ 192.168.xxx.xxx \ DavWWWRoot \ is created without any problem.

You can use this script to run SMBv1:
https://retroroom.pl/reprapy/smbv1.ps1
It should be run with administrator rights, eg run notepad (as administrator) then menu file -> open -> find the downloaded smbv1.ps1 script -> right-click "Run with PowerShell" and enable SMB v1 protocol by following messages in the script.

[smbv1.ps1.txt](https://github.com/bigtreetech/BTT-SD-TF-Cloud-V1.0/files/5683210/smbv1.ps1.txt) (rename file to smbv1.ps1)

Bastlwastl84, 2020-12-16:
> esterday, two ordered WiFI modules came, one SD, the second TF, in the evening I could not get to the SD card connected to the module - I got a connection to the WiFi network, but I did not enter [...]

Smb1.0 was enabled, cause i need it since 2 Years

Seelenkind, 2021-01-24:
> after connecting:
> after i changed the WiFi-Name without a whitespace and updated the Firmeware before update it was not possible that i could connect to wifi. now i got at browser:
> Not found
> URI: / Method: GET
> 
> and at \ "IP-Adress" \DavWWWRoot the error that it can not connect. (doesn't show the double backslashes here)

Ich hatte die selben Probleme anfangs. 
Netzwerk konnte nicht gefunden werden. 
Bis ich den USB Anschluss vom PC getrennt habe und direkt an einen USB Adapter angeschlossen habe. 
Ich denke das Wifi Modul braucht viel mehr Strom als das der USB Port es liefern kann. 
Prompt war die Verbindung da.

MadLad1Clip, 2022-12-10:
So i guess the creator died and there is no fix for this, I just got one and yep not working after connecting to WIFI i cant add the network location.

MadLad1Clip, 2023-07-10:
yeah, pile of shit been trying for months to work out the problem and nope just cant get the windows location added.

fcassia2017, 2024-04-17:
Not sure if anyone else has mentioned this in this thread, but Microsoft has disabled WebDav protocol both on Win10 and Win11 around October 2013 via Windows update.

You need to make sure the web client service is started by invoking services.msc

If that fails, use the windows freeware app WinSCP. Just choose a new connection, select WebDav as protocol in the drop down menu, and enter just the ip address as connection host (no need for /DavWWWRoot).

It just works and implements WebDav internally.

jimkernsjr, 2024-05-03:
> Not sure if anyone else has mentioned this in this thread, but Microsoft has disabled WebDav protocol both on Win10 and Win11 around October 2013 via Windows update.
>
> You need to make sure the web [...]

None of the above worked for me.  What did work:  
Take the card out.  Make a new folder on it called share.
Then use \\ipaddress\share for the link.
I already had SMB1.0 and Web client running - that was NOT the problem.