bigtreetech/BTT-SD-TF-Cloud-V1.0 #23: if it works as wifi it doesn't read me the micro sd

andrub72, 2021-02-02:
ho I configured the micro sd via micro usb to connect to the wifi network and it connects.
When I connect the module to my 3v2 ender 3D printer, it manages to connect to the wifi network, but the printer does not read the micro sd to me.
The mirco sd works because if I insert it into the printer it works.
How can I make the printer read it when it is inside the wifi module

N16htm4r3, 2021-02-15:
I have the same problem, did you find the solution.
-Thanks

Lufibohoc, 2021-02-15:
It has worked perfectly well for me so far. I wanted to put a larger size on the SD card than would fit on it. You gave an error message. It doesn't work now. I tried 3 SD cards. When I put it in, it reads the files. When you connect to wifi, it says you can't see the SD card.
Can I update the wifi unit software?
Thx!

andrub72, 2021-02-16:
I found that with the board of the ender3 it doesn't work, the motherboard is equipped only with the SDIO interface, while this module has only the SP interface!

Lufibohoc, 2021-02-16:
I have an SKR 1.4 Turbo motherboard in my printer. So far, it has worked perfectly. You are now reading the SD card, but only until you are connected to wifi. When connected to wifi it prints that it does not see the sd card.

N16htm4r3, 2021-02-16:
> Ender3 kartının çalışmadığını, anakartın yalnızca SDIO arabirimi ile donatıldığını, bu modülde ise yalnızca SP arabiriminin bulunduğunu gördüm!
I guess I won't use this device as a result. :(
Thx to everyone.

N16htm4r3, 2021-02-16:
> I found that with the board of the ender3 it doesn't work, the motherboard is equipped only with the SDIO interface, while this module has only the SP interface!

Sanırım sonuç olarak bu cihazı kullanmayacağım. :(
Herkese teşekkürler.

N16htm4r3, 2021-02-16:
> I found that with the board of the ender3 it doesn't work, the motherboard is equipped only with the SDIO interface, while this module has only the SP interface!

I guess I will not use this device as a result. :(
Thanks everyone.

andrub72, 2021-02-16:
you cannot ask the module manufacturer for a modification to support the interface as well SDIO !

madjx, 2022-04-01:
I have the same problem, it seems to be really common (see other issues)

Asdf-0123, 2022-08-03:
Hardware error resolution:
https://www.reddit.com/r/SidewinderX1/comments/k4jqt5/wireless_printing_with_a_bigtreetech_tf_cloud_on/

AliasCummings, 2024-11-05:
The SD end plugged into my computer will read an SD inserted into the other end, but it will not read anything plugged into the micro USB port at the other end. Any ideas?