bigtreetech/BTT-SD-TF-Cloud-V1.0 #29: Same file date

Pouet71, 2021-03-02:
Hello, when I transfert a file by wifi, the file modification date is the same 01/01/2000 01:00.
Is there a option to have the good date ?
Thanks

Parkinbot, 2022-02-22:
Found out my device produces the same nasty problem. I am using Windows 8.1. When I copy a file, I first see the correct date, but if I reread the WebDav directory file dates of files copied wirelessly are 01/01/2000. It seems the firmware doesn't write the date at all. Therefore the printer lists new files at the end ...

sarumpaet, 2022-04-12:
There are some forks of ESPWebDav that support this, e.g., https://github.com/MiSU-sw/ESPWebDAV/commit/5a5c5f52a4fd66c74dcd713d1f7205be9e3ad874