bigtreetech/BTT-SD-TF-Cloud-V1.0 #31: if there is always delay ...Troubleshooting

derTino2, 2021-03-09:
I had the problem that the Arduino console always says "Delay ..."

I did everything according to instructions on the Internet:
-Firmware flashed
- Setup.INI on the SD card
-SSID and password in the Setup.ini
-Arduio opened, com selected and the RST button pressed on the ESP12S.

from here on I always had the problem that only "Delay ..." was displayed.

The SOLUTION for my problem was to manually enter the command AT+RST into the console.
And the ESP12S has already connected and has shown me its IP

AlfredBroda, 2021-03-09:
How did the blue light behave? I'm trying to diagnose the same problem, and mine is off for some time, then blue slow blinking. Just sending AT+RST on the Arduino console dos not seem to work. :/

derTino2, 2021-03-10:
The blue light blinks after a while to connect to the router or tries to connect to the router

After connecting, the blue LED was off again and only blinks when the SD card is accessed via the network

DrivebyCoding, 2022-10-18:
Late to the party, but leaving my finding for the next lost traveller:

If the WiFi connection is not established, in my case because I had the wrong casing on the SSID name, the serial output will say nothing, just hang a at "Delay ..." forever. After fixing the SSID and pressing reset it printed it's IP as shown in the documentation.