bigtreetech/BTT-SD-TF-Cloud-V1.0 #44: begin SD failed

csanz, 2023-05-24:
I'm getting the following error 

```
begin SD failed
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  feefeffe feefeffe 3fff0554 4020233e  
3fff1560:  feefeffe feefeffe feefeffe 40208569  
3fff1570:  3fffdad0 00000000 3fff0554 40209074  
3fff1580:  feefeffe feefeffe 3fff0570 40100a28  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
115200
```

[image]

I'm using the same SSID and PWD from the instructions 

SSID=biqu
PASSWORD=biqu2019

Connecting with a Mac computer. 
Any help is highly appreciated.

GabrielSerealia, 2023-06-02:
Having the same issue as well
[image]

timgreen, 2023-08-31:
Maybe try another card. 

I found one of my old 8GB card (no brand) works.

jonboy1414, 2024-01-12:
i am having same prob. been at it for 2 days try 4 diffrent cards all dif size