bigtreetech/BTT-SD-TF-Cloud-V1.0 #46: open file failed

DannyJBill, 2023-05-27:
I'm getting the following error
'Delay...
open file failed
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  feefeffe feefeffe 3fff0554 4020233e  
'

DannyJBill, 2023-05-27:
Delay...
open file failed
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  feefeffe feefeffe 3fff0554 4020233e

DannyJBill, 2023-05-27:
'Delay...
open file failed
Fatal exception 28(LoadProhibitedCause):
epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (28):
epc1=0x402093d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff13a0 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff1540:  feef0001 3ffefce8 3fff0554 402022fd  
3fff1550:  feefeffe feefeffe 3fff0554 4020233e  
3fff1560:  feefeffe feefeffe feefeffe 40208569  
3fff1570:  3fffdad0 00000000 3fff0554 40209074  
3fff1580:  feefeffe feefeffe 3fff0570 40100a28  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8'

MrWillieMatt, 2023-05-29:
> I'm getting the following error 'Delay... open file failed Fatal exception 28(LoadProhibitedCause): epc1=0x402093d4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000
>
> Exception [...]

Same issue on mine today

MadLad1Clip, 2023-06-12:
make the new file a RTF file not a note card.

KajPedersen, 2024-04-12:
I have been struggling with the same issue, for a week now, trying everything suggested here and there.
However, issue seems solved today.
Downloaded Notepad++ and saved the text document as a "MS ini file", and there it was. Just my day today

djjohnson13p, 2024-07-03:
> I have been struggling with the same issue, for a week now, trying everything suggested here and there. However, issue seems solved today. Downloaded Notepad++ and saved the text document as a "MS ini file", and there it was. Just my day today

Thank you been racking my brain for hours over this such an easy fix!